jankolkmeier/xbee-api #21: Mysterious buffer_reader functionn

jankolkmeier, 2015-01-15:
@joshperry , can you explain [this line](https://github.com/jankolkmeier/xbee-api/blob/0cd17537f7a4157969cf3f628e06c57a9000e0a5/lib/frame-parser.js#L126) in your earlier commit:

```javascript
frame.nodeIdentifier = reader.nextSze('ascii');
```

Nowhere I can find a reference to a ```nextSze()``` function in the buffer-reader module. Also it didn't seem like you extended the buffer-reader yourself in an earlier commit. 

The really weird thing though is that the code passed several nodeunit tests with this line inside - except for today, when it suddenly turned up!

I added the function myself in 7d672e8 to fix this, but I'm still kind of baffled about this.

joshperry, 2015-01-15:
@jankolkmeier, `nextSze` means "next string, zero terminated" (old hungarian notation reference). It's a function that I added to the two libs in a PR; the maintainer of `buffer-writer` decided (rightly so) to use `nextStringZero` instead.

You never really know when someone is going to take a PR, sometimes when little libs like this haven't been updated in years, the author has moved onto other things, sometimes not even related to programming, that's why I added our git repos as dependencies. This looks like that's what happened with the `buffer-reader` lib.

It looks like the issue you're seeing is because the deps from npm are in the `devDependencies` while the deps from my git repos are in the `dependencies` so you weren't getting the rev with my changes.

I have a couple of those things fixed already, I just wanted to fix a couple other things before I sent another PR. Having git repos as deps isn't really a great idea because you can't install the module unless you have git on your system. I'll fork `buffer-reader` and push to npm and get you a PR for that.

Apologies for the sloppy testing and breaking the last rev, I really need to stay more diligent with that.

jankolkmeier, 2015-01-15:
Ah, makes perfect sense now! I completely overlooked the option of having git deps! 

As for that last rec, I also should have made a test case for the nextFrameId function, and verified functionality beyond just running the tests :)

So thanks a lot for the explanation and your work so far!

closed 2015-01-15 by jankolkmeier