jankolkmeier/xbee-api #27: Information

beinnova, 2015-04-20:
Hi, 
Firstly complimet for the great job!
Secondoly I have one question for you: it's possible to make some personals frame builder/parser to send data with my personal form?

jankolkmeier, 2015-04-20:
Thenks for the coplimet.

I'm not sure what you want to achieve. A HTML form? You want to use xbee-api in the backend of your website? You want to make a web interface?

Whatever it is, the answer is probably yes.

beinnova, 2015-04-20:
I'm sorry, I wrote the last post in a very bad english :smile:  . I mean if it's possible to make personal frame builder for send parsonal frame with personal parameters. Like Digi Sensors.

xianlin, 2015-04-21:
It is possible. XBee is just a carrier to send your frame content. You can define your own frame with your own parameters. 

One example is Libelium's waspmote, they defined and used their own frame. You can take a look into their manuals for more details.

beinnova, 2015-04-22:
​
Ok, i have find the solution. Now I can parse my personal frame. Thanks a
lot.

​Regards

Giorgio​

closed 2015-04-29 by jankolkmeier