jankolkmeier/xbee-api #30: Parse Frame Buffer Too Small

traviscosgrave, 2015-04-26:
I noticed that the `Buffer` for parsing raw frames is 128 octets. I think this is too small. With a small amount of text data I've been able to blow 128 out of the water and I receive a `RangeError: index out of range`.

From what I can tell, ATNP indicates the maximum payload size. On my XBee Pro S2Bs, I get `FF` for the NP parameter. I loosely tested this in XCTU by sending `10` packets with their payload filled with '1'. When the total frame length hit `01 0E` (270) I received Transmit Status `8B` messages with Delivery Status `74` (Data payload too large). The maximum frame length I could send was 269. When the 269 length frame was received, the length was `01 11` (273). Add 3 to that for the start and legnth header and that gives 276 for the required buffer size.

I am using API=1 AO=1. I'm not sure how the length changes with escaped frames.

Because I'm pretty new at this, I'm not sure of the possible side effects of upping the `parseRaw ` buffer size. I imagine it's additional memory allocation? I haven't found a great definitive answer on the absolute maximum frame size for ZigBee, 802.15.4 or the like. However, I'm pretty certain that a `Buffer` size of 512 would probably cover all needs and likely be excessive. Maybe a default of 512 with the option to configure it yourself as an option would work for those with truely memory bound inplementations.

Thoughts?

jankolkmeier, 2015-04-26:
I thought I based this number on the specs, but what you describe seems clear. I'm confident that there will be no side effects changing to 512.

I assume you tried this out, and it fixed your problems?

traviscosgrave, 2015-04-26:
Yes, it did fix the problem.

My primary curiosity about the max frame size would is motivated by the
desire to set the buffer to the correct size (absolute max frame length),
and write a test that demonstrates the buffer's ability to handle that
frame size and throw an easier to read error if something larger is
discovered. Although that shouldn't ever be the case. The sending device
will get a message indicating the frame payload is too large, but a
receiving device never sees the frame. Currently the entire process halts
even though a valid frame is received.

If that change and test would be useful, I'm happy to make the PR. If you
think just handling a large buffer than can ever be encountered is fine,
then the test is probably not necessary.

jankolkmeier, 2015-04-29:
I'd be glad to accept a PR! If you are currently facing a scenario that could serve as a test for this, writing a test should be easy. But if you have to artificially build the test from scratch, it seems like a little too much work :)

yesakmac, 2015-11-06:
Nearly a complete newbie with the xbee protocol here, but I believe I'm having the same issue with buffer size. The frame is able to be viewed in the console of the sender in it's entirety, but when I attempt to log it to the console of the recipient it shows the following: 

>> { type: 144,
  remote64: '0013a20040d87f61',
  remote16: 'fffe',
  receiveOptions: 194,
  data: <Buffer 3c 3d 3e 80 01 23 34 30 30 35 37 37 32 36 33 23 57 41 53 50 4d 4f 54 45 5f 47 50 53 23 32 38 23 53 54 52 3a 4e 6f 20 47 50 53 4e 6f 20 47 50 53 4e 6f ... > }

Is this indicative of this buffer issue, and if so, how do I fix it? 

Thanks!

jankolkmeier, 2015-11-08:
First off, I think this issue is a duplicate of #33 , which got resolved.

@traviscosgrave can you confirm? It seems to me that we can now reduce the buffer size again closer to the theoretical maximum?

@yesakmac: To me it doesn't seem related. I'm not even sure what issue you are are describing here? Do you get a RangeError exception? On what end? What is the code?

jankolkmeier, 2015-11-08:
(the correct issue with the fix is #34 )

jankolkmeier, 2016-01-08:
merged fix in 0.4.0

closed 2016-01-08 by jankolkmeier