jankolkmeier/xbee-api #31: Detect when a new Xbee Device joins the network

vertangelx, 2015-05-03:
How can we call a function when the XBee (Series 2, Coordinator API) detects that another XBee have joined the network?

jankolkmeier, 2015-05-03:
xbee-api does not (and will not) provide such functions for you. this is a matter of configuring your xbee's correctly.

You have to enable the JN setting on the router/end-device and set the AO setting to 0 on the coordinator (see [p.163 and 166](http://ftp1.digi.com/support/documentation/90000976_W.pdf) for more info on JN and AO).

Devices joining the network with JN=1 will broadcast a join notification packet. Your coordinator will receive this, and if you are already listening to the UART at that moment and let xbee-api parse the message, xbee-api will emit a [Node Identification Indicator (0x95)](https://github.com/jankolkmeier/xbee-api#0x95-node-identification-indicator-ao0-znet-zigbee) frame object.

If you want to know about nodes that have joined before you started listening to the UART, you'll have to make an ATND request.

closed 2015-05-12 by jankolkmeier

tomlogic, 2018-07-02:
FYI, documentation link above (p.163 and 166) was dead.  Current documentation available at https://www.digi.com/resources/documentation/digidocs/PDFs/90000976.pdf.