jankolkmeier/xbee-api #32: Support for Xbee Serie 1 Digimesh protocol

giorgiocanale, 2015-05-12:
Hi, is there any chance you might support also the frames for digimesh protocol?
Digimesh allows the modules to create a self healing full-mesh network with xbee series 1 modules from DIGI. There is some python support, but so far no node that I found of support this protocol.
It should suffice to support some more frames in my understanding...

Thanks

jankolkmeier, 2015-05-12:
I do not have the time to do this, but yes - seems like writing parsers for the respective frames is all you need. So feel free too send a pull request ;)

docwelch, 2016-03-08:
I looked at what it would take and it seems that only two parsers need to be written - one for Route Information Packet (0x8D) and Aggregate Update Addressing (0x8E). I forked this project and created a DigiMesh branch on my Github account, docwelch. I made a couple of additions to constants.js and to frame_parser.js. I do not have a Digimesh network set up at the moment to test this, unfortunately. If @giorgiocanale would like to test, that would be great. If it is working, I will submit a pull request.

giorgiocanale, 2016-07-13:
Hi, I'm having hard time to make it works, it keeps giving me checksum
error.
I'm not sure this depends on your modifications or else from some version
mismatch on required library.
I'll try to get to the bottom of it, and let you know

Thanks

jankolkmeier, 2017-07-14:
@giorgiocanale did you ever figure this out? Checksum mismatch happens long before the parsers @docwelch wrote are being used. Sounds more like the module wasn't configured properly (wrong api mode, baud rate, etc.?).

martin9115, 2021-07-16:
Hi, does anyone solve the digimesh problem?