jankolkmeier/xbee-api #35: Command Data parser for Network Identifier with Network Discovery Command

jonhardman, 2015-12-21:
Hi,
I have been reviewing the xbee api for a while now and it is fantastic ! :-) 
However I have found something I seem to be unable to do with the api, I am running a simple xbee setup with one coordinator and one router in API mode, I am successfully sending a ND command, and the response is received. However what I am unable to achieve through the api is to be able to parse the Network Identifier value that is received in this packet in command data:

{ type: 151,
  id: 1,
  remote64: '0013a20040cc03bc',
  remote16: '0000',
  command: 'ND',
  commandStatus: 0,
  commandData: <Buffer 00 00 00 13 a2 00 40 cc 03 bc 20 48 31 00 ff fe 00 00 c1 05 10 1e> 
}

Is this possible with the api or is it not available yet?

Thank you,

Jon

jankolkmeier, 2016-01-04:
Hey Jon,

Sorry for taking so long to get back to you!

We can parse that buffer with the parseNodeIdentificationPayload function (see lib/frame-parser.js).
This is automatically done for regular AT requests, but not yet for remote ones. I'll write a quick patch that you can try out that should leave you with a nodeIdentification property in that packet, that contains the parsed information.

Best,
Jan

jonhardman, 2016-01-04:
Hi Jan,

That has worked perfectly thank you, I can see the node identifier within the node identification packet now.
Will this be something that gets merged into the master at somepoint? I can imagine it being useful for others.

Jon

jankolkmeier, 2016-01-04:
Seems like a safe merge that shouldn't break anything for people. I will
merge it later today!

jankolkmeier, 2016-01-08:
merged fix in 0.4.0

closed 2016-01-08 by jankolkmeier