jankolkmeier/xbee-api #37: frame_parser.ParseIOSamplePayload throws assertion for "IS" command if no IO configured

abythell, 2016-03-12:
Consider the following remote AT request:
```
{
  "type":23,
  "destination64",
  "0013a200408de1fe",
  "command":"IS",
  "commandParameter": []
}
```

If the remote XBee radio has IO lines enabled (eg. D1=3), the response will be an IO Sample packet, which ParseIoSamplePayload() parses happily.

But, If the remote XBee radio does not have any IO lines enabled, it still responds with an IO Sample packet, but with commandStatus=1.  Further, the frame is shorter and does not contain the number of samples, which causes:
```
AssertionError: Out of Original Buffer's Boundary
    at BufferReader.MAKE_NEXT_READER.BufferReader.(anonymous function) [as nextUInt8] (/home/user/node_modules/xbee-api/node_modules/buffer-reader/index.    at Object.frame_parser.ParseIOSamplePayload (/home/user/node_modules/xbee-api/lib/frame-parser.js:207:29)
    at Object.frame_parser.(anonymous function) (/home/user/node_modules/xbee-api/lib/frame-parser.js:109:18)
    at XBeeAPI.parseFrame (/home/user/node_modules/xbee-api/lib/xbee-api.js:113:27)
    at XBeeAPI.parseRaw (/home/user/node_modules/xbee-api/lib/xbee-api.js:203:26)
    at XBeeAPI.<anonymous> (/home/user/node_modules/xbee-api/lib/xbee-api.js:133:10)
```

I think the fix is to add:

```
if (frame.commandStatus !== 0) return;
```

at frame-parser.js, line 206, but I need to test and confirm this.

jankolkmeier, 2016-03-13:
Thanks for spotting this! See the patch I submitted. I'm also checking if the frame.commandStatus property is set. The ParseIOSamplePayload function is also used to parse 0x92 packets (ZigBee IO Data Sample Rx Indicator), which of course don't have the commandStatus property (it's AT command/response related).

Could you check if this works for your problem?

abythell, 2016-03-21:
Forgive the lateness of my reply.  Yes, with this patch IS responses from remote radios with and without IO lines enabled are parsed correctly.

jankolkmeier, 2016-03-22:
No worries, thanks a lot!

closed 2016-03-22 by jankolkmeier