jankolkmeier/xbee-api #39: ATND command error response causes exception

ntokos, 2016-03-23:
Hi all,

I'm using the xbee API lib in a beaglebone with XBee module (S2, firmware version 21A7) attached (experimenting on a ZB/IP gateway).
I'm using the ATND<node id string> command to resolve the node identifier of a ZB node. Sometimes the command causes, after some secs, an exception like the following:

```
assert.js:93
  throw new assert.AssertionError({
        ^
AssertionError: Out of Original Buffer's Boundary
    at BufferReader.nextString (/usr/local/lib/node_modules/xbee-api/node_modules/buffer-reader/index.js:57:5)
    at Object.frame_parser.parseNodeIdentificationPayload (/usr/local/lib/node_modules/xbee-api/lib/frame-parser.js:190:27)
    at Object.frame_parser.(anonymous function) [as 136] (/usr/local/lib/node_modules/xbee-api/lib/frame-parser.js:96:18)
    at XBeeAPI.parseFrame (/usr/local/lib/node_modules/xbee-api/lib/xbee-api.js:113:27)
    at XBeeAPI.parseRaw (/usr/local/lib/node_modules/xbee-api/lib/xbee-api.js:203:26)
    at XBeeAPI.rawParser (/usr/local/lib/node_modules/xbee-api/lib/xbee-api.js:133:10)
    at Object.opts.dataCallback (/usr/local/lib/node_modules/serialport/serialport.js:181:12)
    at SerialPort._emitData (/usr/local/lib/node_modules/serialport/serialport.js:396:20)
    at afterRead (/usr/local/lib/node_modules/serialport/serialport.js:372:20)
    at /usr/local/lib/node_modules/serialport/serialport.js:388:9
```

The problem seems to be that If node discovery fails (for example the node doesn't respond to discovery), the XBee module produces an ATND command response with ERROR (0x01) which is not handled in the frame parser.

A possible fix would be a small change in function _frame_parser[C.FRAME_TYPE.AT_COMMAND_RESPONSE]_ (lib file _frame-parser.js_):

```
frame_parser[C.FRAME_TYPE.AT_COMMAND_RESPONSE] = function(frame, reader) {
  frame.id = reader.nextUInt8();
  frame.command = reader.nextString(2, 'ascii');
  frame.commandStatus = reader.nextUInt8();
  if (frame.command === "ND") {
    frame.nodeIdentification = {};
    frame_parser.parseNodeIdentificationPayload(frame.nodeIdentification, reader);
  } else {
    frame.commandData = reader.nextAll();
  }
};
```

where we replace the if statement (line 94):
`  if (frame.command === "ND") {`
to:
`  if ((frame.command === "ND") && (frame.commandStatus == C.COMMAND_STATUS.OK)) {`

This will prevent a call to frame_parser.parseNodeIdentificationPayload in case of a ATND command error.

jankolkmeier, 2016-03-23:
Thank you a lot for reporting this!

Another option could be to check in parseNodeIdentificationPayload if there is any data left in the reader, but I prefer your fix by far.

I guess the same fix should then also be added in `frame_parser[C.FRAME_TYPE.REMOTE_COMMAND_RESPONSE]`

I will write tests and a patch later this week (or, if you feel like it, you can also do a PR :) ).

ntokos, 2016-03-23:
Thanks for the immediate response.

I agree, a similar fix should be added in the remote command response frame_parser function. Only for reasons of completeness though. I can't think of a good reason why would someone need to do discovery for a specific node ID string via a remote ATND command, he can just ask the "local" XBee to do that discovery :).

I'm a complete new github user, so that PR (=pull request?) might take some time :)

jankolkmeier, 2016-03-23:
Don't worry, If you want to learn it, just give it a shot :)

jankolkmeier, 2016-03-24:
Thanks & good job! Updated the package!

closed 2016-03-24 by jankolkmeier