jankolkmeier/xbee-api #41: Really Basic Question - (Hoping for a response by tomorrow afternoon, if possible!)

KyleKing, 2016-04-13:
Hi, I'm working on a basic xbee-api/serialport device. I used the sample code from the README:
```js
// Something we might receive from an XBee...
var raw_frame = new Buffer([
    0x7E, 0x00, 0x13, 0x97, 0x55, 0x00, 0x13, 0xA2, 0x00, 0x40, 0x52, 0x2B,
    0xAA, 0x7D, 0x84, 0x53, 0x4C, 0x00, 0x40, 0x52, 0x2B, 0xAA, 0xF0
]);

console.log(xbeeAPI.parseFrame(raw_frame));
// { type: 151,
//   id: 85,
//   remote64: '0013a20040522baa',
//   remote16: '7d84',
//   command: 'SL',
//   commandStatus: 0,
//   commandData: [ 64, 82, 43, 170 ] }
```

However, I only get this when using the `xbeeAPI.parseFrame(raw_frame)`:
```txt
{ type: 151,
  id: 85,
  remote64: '0013a20040522baa',
  remote16: '7d84',
  command: 'SL',
  commandStatus: 0,
  commandData: <Buffer 40 52 2b aa> }
```

My question is what to do with the data field: `<Buffer 40 52 2b aa>`? In the example, it should be: `[ 64, 82, 43, 170 ]`.

I tried `frame_obj.data.toString();` and I get `@R+�`? Do you have any advice on how to read the data field?

ntokos, 2016-04-13:
Hello,
using .toString() method is equivalent to .toString('ascii'), which means you convert your hexadecimal number to the ASCII char that corresponds to the value. You should use .toString('hex').

KyleKing, 2016-04-13:
Thanks @ntokos, you were definitely right that I was using .toString() wrong, but using .toString('hex') incorrectly returns: `40522baa`?

KyleKing, 2016-04-13:
Solved it!

```js
function hex2d(hex) {
  var decimal = [];
  for (var i = 0; i < hex.length; i += 2) {
    decimal.push(parseInt(hex.substr(i, 2), 16));
  }
  console.log(decimal);
  return decimal;
}

hex2d(data.toString('hex'));
// returns: [ 64, 82, 43, 170 ]
```

[Original example code for hex → ascii](http://stackoverflow.com/a/3745677/3219667)

closed 2016-04-13 by KyleKing