jankolkmeier/xbee-api #46: Wrong UNKNOWN_64 constant value

marianofino, 2016-06-08:
Hi!

I've found another small bug.. According to ZigBee's (and Xbee) specification the unknown 64bit address is `0xFFFFFFFFFFFFFFFF`. In the module, the `UNKNOWN_64` constant is defined as `0x000000000000FFFF`, which actually is Xbee's broadcast address (I think in Zigbee's specs broadcast is defined in the network layer, so this doesn't count).

It may seem that nothing changes, but when using Xbee it actually reduces traffic a lot (thing I am trying to achieve), when creating frames without the `destination64` attribute.

Bye!

marianofino, 2016-06-08:
I've been debugging this, and it is not working as I expected. Although as far as I understand this should work (more info about broadcast and unicast in Xbee here: https://docs.digi.com/display/XBeeZigBeeMeshKit/Data+transmission), in my tests the frames with unknown 64bit addresses are not being sent to the neighbours.

I'll update you if I find more info about it..

jankolkmeier, 2016-06-09:
Very interesting, indeed. I'm curious what you will find.