jankolkmeier/xbee-api #47: XBEE 868LP ADC

Flo1991, 2016-08-18:
Hi,
the xbee-api is really good for interfacing XBEEs. But as I tried a remote adc I got strange values
using XBEE 868LP. A look up in the API shows the reason: The adc values are no raw values...
If you look at frame-parser.js the adc samples are converted like this:
 frame.analogSamples[C.ANALOG_CHANNELS.MASK[abit][0]] = Math.round((valA * 1200) / 1023);

That is nice for different XBEEs, but not for the 868LP. The default reference voltage is 2,5 V and you also can choose 1,25 V (no 1,2 V reference available). So I suggest not to convert the adc value into a voltage value and return the raw value.

A workaround is to do the math back:
valA *1023/1200

Flo1991

jankolkmeier, 2016-08-19:
Thanks for pointing this out. I never looked at the 868LP.
While you have a point in that we should not just compute that value, as it may depend on implementation. I'd rather have it configurable (i.e. pass the type of module in the XBeeAPI constructor options), or return the raw value and keep valA to not break existing implementations?

Flo1991, 2016-08-19:
A solution may be returning the raw value in addition, something like:
analogSamples: { AD0: YYY, AD0raw: XXX}

Flo1991

math3os, 2016-11-04:
My module hardware is a Xbee-PRO- S3B and I can set my own Vref for analog input. So i suggest to add an option to disable conversion AND/OR an option to set VREF at instantiation. May be i will do it now.

math3os, 2016-11-04:
pull request : #50

closed 2016-11-07 by jankolkmeier