jankolkmeier/xbee-api #52: Unit Testing XBEE - mock

jonhardman, 2016-11-23:
Hi,
I have been using this api for a while now and it is great :-)

I was wondering if anyone has figured out how to mock out a xbee packet to simulate a number of packets arriving through the serial channel.

Thanks in advance!

Jon

101100, 2016-11-23:
I mocked part of the library for tests in [xbee-rx](https://github.com/101100/xbee-rx).  The mock is [here](https://github.com/101100/xbee-rx/blob/master/test/mock-xbee-api.js) and one instance of its use is [here](https://github.com/101100/xbee-rx/blob/master/test/remoteCommand-tests.js#L330-L335).

jonhardman, 2016-11-28:
Thanks for the advice, will look into it :-). I've only used the xbee-api so far and not the xbee-rx version, would you say there is value in moving to the rx version?

101100, 2016-11-28:
@jonhardman `xbee-rx` builds upon `xbee-api` and takes care of the serial port connection and pairing off API calls with their responses.  It is based upon [RxJS](https://github.com/Reactive-Extensions/RxJS) and provides streams for command results, messages and responses.  The simpler `xbee-promise` library handles local and remote commands with promises but doesn't handle incoming messages (since the promise model only handles single events, not multiple events in a row like a sequence of incoming messages).