jankolkmeier/xbee-api #53: Suuport Node v4 (and possibly older)

101100, 2016-11-27:
The latest version (`0.4.6`) no longer supports Node versions before v6.  This appears to only be the case due to the use of the new Buffer APIs.

It would be nice if this library supported older versions of Node.js.  This is helpful for small board computers where the latest version is not always available.

The [`safe-buffer`](https://github.com/feross/safe-buffer) library provides access to the new APIs for older version of Node and has no other dependencies.  Would you accept a change to use that?

jankolkmeier, 2016-11-28:
First of all, sorry if this change caused any inconvenience for you. Maybe this change would've better been denoted by a major version.

Although I prefer to keep as few deps as possible, I think this is a very reasonable change. And I am certainly interested in supporting small boards / IoT devices. 

Will you make a pull request?

jankolkmeier, 2016-11-28:
Additionally, maybe it would be good to update the Travis unit tests to include multiple versions (before and after 0.4).

PR #54 (2016-12-02): Support old versions of Node.js
This updates the library to use `safe-buffer` which is a tiny library that mocks the new Buffer APIs found in v5.x and beyond for use in older versions of Node.js.

@jankolkmeier I didn't update any of the examples or tests, just the library itself.  Let me know if you'd like me to change that.

Closes #53
closed 2016-12-02 by jankolkmeier