jankolkmeier/xbee-api #57: Send an ACK when coordinator receive data from any xbee

fxsh1301, 2017-02-22:
Hello everyone!
The code work greatly so far, but I'm trying to figure out how to make a ack.
What I want to do is make coordinator with Pi send a "ACK" when it receive data from any xbee.
The code below allows me to receive all of data :
`xbeeAPI.on("frame_object", function (frame) {
    console.log(">>", frame);
    var d= new Date();
    var yyyy =d.getFullYear()
    var MM = (d.getMonth()+1<10 ? '0' : '')+(d.getMonth()+1);
    var dd = (d.getDate()<10 ? '0' : '')+d.getDate();
    var h = (d.getHours()<10 ? '0' : '')+d.getHours();
    var m = (d.getMinutes()<10 ? '0' : '')+d.getMinutes();
    var s = (d.getSeconds()<10 ? '0' : '')+d.getSeconds(); 
    console.log( ""+h+"-"+m+"-"+s+"-"+dd+"-"+MM+"-"+yyyy);
    recadd = frame.remote64;
    console.log(recadd);
    });`
The variable "recadd" is the remote 64bit address.
When i put the code inside it will send several time in short time:
`serialport.on("open", function () { //send the ack back to xbee which recieve
              xbeeAPI.on("frame_object", function (frame) {
              var recadd = frame.remote64;  
                var frame_obj = {
                    type: 0x10, 
                    id: 0x01, 
                    destination64:recadd ,
                    broadcastRadius: 0x00,
                    options: 0x00,
                   data: "ack"
                      }; 
                serialport.write(xbeeAPI.buildFrame(frame_obj));
                console.log('Sent to serial port.');
                });
    });`
Is there anyway to achieve that ?

jankolkmeier, 2017-02-22:
Sorry, this is clearly more of a lack of general understanding of how the xbee API mode works, and not a problem with this module. You'd be better of asking this on stackoverflow or the digikey forums.

Still, I'll point you in the right direction: One problem with your code is that just sent the "ack" in response to **any API frame**. This is because ```on("frame_object", ...)``` emits all types API frames - not just data frame (such as the 0x90 frames), but also frames such as transmit status (i.e. 0x8B), which you get back whenever you make a transmit request (such as 0x10). So this code basically starts an infinite loop.

closed 2017-02-22 by jankolkmeier

fxsh1301, 2017-02-22:
Thanks jankolkmeier.
Because I need to know which xbee just send data to here, so use on("frame_object", ...) to get the remote64 to know the 64 bit address, and then send the ack back.
So how can I implement do something when coordinator receive any api frames?
I'm appreciated for your reply.