jankolkmeier/xbee-api #59: Parsing fails if START_BYTE occurs in radio address, using API Mode 1

geekoftheweek, 2017-03-18:
One of my radios had 0x7E in its 64-bit address, and parsing failed.  I've attempted to fix this in #58 but may have missed something.

jankolkmeier, 2017-03-21:
That's a major problem! thank you so much for catching this and providing a fix!

That extra check is certainly preferred behavior and the added test case is good to have and it demonstrates that as good as possible that the parsing returns to the proper state when finishing a packet.

As far as I can tell it should work as intended also in practice. Have you tested this in practice yet?

I also am thinking of maybe changing that line in the future to this slightly more complex one:
```javascript
if ((S.waiting || (this.options.api_mode === 2 && !S.escape_next)) && S.b === C.START_BYTE) {
    // ...
}
```

This way we get the full benefit of using the AP=2 mode: If we see (without escape) the 0x7e byte in AP=2 mode, it MUST mean that it is a start byte and we can immediately reset to a state accepting a new package.

Whereas in AP=1 mode, as you noticed, it might just be the coincidental 0x7e in the address or in some other payload... 
And if we're indeed parsing a malformed package, there is no way of knowing until we parse more bytes that we expect to parse before going back to a state where we accept the start byte again (i.e. S.waiting=true)... and in the meantime, we might fail to parse backages properly (whereas we can recover better in AP=2 mode).

(A good reason to try using AP=2 :) )

geekoftheweek, 2017-03-21:
That revised line makes perfect sense -- wondering about the consequences of my change on AP=2 has been nagging at me over the past few days.  I don't have experience working with API mode 2, but your proposed change seems reasonable and correct to me!

kevz93, 2017-05-26:
Can this explain why I'm not able to send anything above 4 bytes?

 The Rx seems to not give any response also.

jankolkmeier, 2017-05-29:
The fix for the issue discussed here has not been merged yet.
I am not 100% convinced that it won't break something else, so I am cautious.

It could only be related to whatever you are describing, if you are trying to send a message with value 0x7e at byte 5. Please make a separate issue otherwise (and give a detailed description of your setup and minimal code to reproduce the issue, and what you think it should do instead).

closed 2017-07-14 by jankolkmeier