jankolkmeier/xbee-api #60: sending frames while listening frames

fxsh1301, 2017-03-27:
Hello I'm using xbee-api and it work greatly.
I'm using xbeeAPI.on to listening and send another frame back after received.
But when I want to send other frames out of xbeeAPI.on , it can't be sent because of the port will be lock.
Also ,i can't put it in xbeeAPI.on, because that will run only when frames coming.
How can I run another function without xbeeAPI and keep original function run?
Thanks.

jankolkmeier, 2017-03-27:
You will have to provide example code, explain specifically what you want it to achieve and explain exactly it does instead. Based on your description there is no way of telling what you (are trying to) do.

fxsh1301, 2017-03-28:
var util = require('util');
var SerialPort = require('serialport');
var xbee_api = require('xbee-api');

var C = xbee_api.constants;

var xbeeAPI = new xbee_api.XBeeAPI({
    api_mode: 2
});

var serialport = new SerialPort("/dev/ttyS0", {
    baudrate: 9600,
    parser: xbeeAPI.rawParser()
});

  

var minutes = 5, the_interval = minutes * 60 * 1000;
setInterval(function() {
console.log("5min check");
  //The port will be lock if I send frame from here.
}, the_interval);

// All frames parsed by the XBee will be emitted here
xbeeAPI.on("frame_object", function (frame) {
// The frame can't be send every 5 minutes in here, only when frames come in. 
var d= new Date();
var yyyy =d.getFullYear()
var MM = (d.getMonth()+1<10 ? '0' : '')+(d.getMonth()+1);
var dd = (d.getDate()<10 ? '0' : '')+d.getDate();
var h = (d.getHours()<10 ? '0' : '')+d.getHours();
var m = (d.getMinutes()<10 ? '0' : '')+d.getMinutes();
var s = (d.getSeconds()<10 ? '0' : '')+d.getSeconds(); 
 
    console.log(">>", frame);
    console.log( ""+h+"-"+m+"-"+s+"-"+dd+"-"+MM+"-"+yyyy);
   var recadd = frame.remote64; //source address
    console.log(recadd);
            if(recadd!=null){
              recadd = frame.remote64;
              var frame_obj_ack = {
                    type: 0x10,
                    id: 0x01,
                    destination64: recadd ,
                    broadcastRadius: 0x00,
                    options: 0x00,
                    data: ""+h+m+s+dd+MM+yyyy
                 };
              serialport.write(xbeeAPI.buildFrame(frame_obj_ack));
          //    console.log('Sent to serial port.');
              console.log(recadd);
              
            }         
    });

jankolkmeier, 2017-03-28:
What is the error that is actually produced? "The port will lock" sounds more like your interpretation of what is happening.

One problem could be that you start the interval before you know whether the serial port is already open. See how to do stuff on('open') under "opening a port" [here](https://github.com/EmergingTechnologyAdvisors/node-serialport) (i.e.: start the interval from the on open callback of serialport).

Also, your code does not check for the frame type in on('frame_object'). That's asking for problems, as the code will also be run whenever a status frame (i.e. ack sending a frame) or similar is being produced by the xbee.

PS: A tip for the future: you could communicate your problem better by producing the minimal necessary code to reproduce the error. This is unreadable (also: learn how to format code on github) and has a bunch of unnecessary stuff (like the whole date formatting nonsense).

closed 2017-07-14 by jankolkmeier