jankolkmeier/xbee-api #61: XBee S2C didgimesh support

brituj, 2017-07-14:
Hi, is there any chance of you guys adding digmesh protocol support?

jankolkmeier, 2017-07-14:
I have no interest in doing it, but feel free to contribute!

jankolkmeier, 2017-07-14:
As a note, if you look at the [api frames used by s2c digimesh](https://www.digi.com/resources/documentation/Digidocs/90001506/Default.htm#reference/r_api_frame_overview.htm%3FTocPath%3DOperate%2520in%2520API%2520mode%7CAPI%2520frame%2520overview%7C_____0), there is only a couple of unimplemented frames for the frame parser (that you may not even need):

- [0x8d: Trace Route or NACK Data](https://www.digi.com/resources/documentation/Digidocs/90001506/Default.htm#reference/r_frame_0x8d.htm%3FTocPath%3DOperate%2520in%2520API%2520mode%7CAPI%2520frame%2520overview%7C_____10)
- [0x8e: Aggregate Addressing Update](https://www.digi.com/resources/documentation/Digidocs/90001506/Default.htm#reference/r_frame_0x8e.htm%3FTocPath%3DOperate%2520in%2520API%2520mode%7CAPI%2520frame%2520overview%7C_____11)
- [0xA0: Over-the-Air Firmware Update Status (ZigBee)](https://www.digi.com/resources/documentation/Digidocs/90001506/Default.htm#reference/r_frame_0xa0.htm%3FTocPath%3DOperate%2520in%2520API%2520mode%7CAPI%2520frame%2520overview%7C_____17)

Implementing any of these is done by:

 - Adding the constants for the frame types (and possible response values) to [constants.js](https://github.com/jankolkmeier/xbee-api/blob/master/lib/constants.js)
 - Writing a test for the frame to be implemented in [xbee-api_test.js](https://github.com/jankolkmeier/xbee-api/blob/master/test/xbee-api_test.js). For these packets, all being packets from the module, add the tests under `'API Frame Parsing'`. The bytes in `rawFrame` can be inspired by the examples given on the digi website, and in the frame callback, you test if the object properties are what you expect them to be. 
 - Implementing the in [frame-parser.js](https://github.com/jankolkmeier/xbee-api/blob/master/lib/frame-parser.js) (as you can see, the to be parsed frame starts after the frame type byte, and does not contain the checksum byte at the end.

jankolkmeier, 2017-07-14:
And finally, if you look at the last four commits in [this branch](https://github.com/docwelch/xbee-api/commits/DigiMesh) of @docwelch, you will find that somebody already did the work. but according to issue #32, he just couldn't test it, yet.