jankolkmeier/xbee-api #62: Serialport 5 incompatibility

docwelch, 2017-08-04:
Serialport 5 has changed to streams. As noted here, [Serialport 5 upgrade](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/92c7dd4bfd2c05f2fe39399a94eba0c0ebb27434/UPGRADE_GUIDE.md), custom parsers will have to be upgraded. Is this something you will be supporting?

Thanks

jankolkmeier, 2017-08-06:
Sure, I mean, that's the way to go.

I guess the solution would be to make the xbee-api class stream-compatible? Sortoflike in [xbee-stream.js](https://github.com/jankolkmeier/xbee-stream/blob/master/lib/xbee-stream.js) I guess? I don't have the time atm to look at this too closely...

jankolkmeier, 2017-11-29:
Added in v0.6.0. The example in the README is updated to show the stream interface.

closed 2017-11-29 by jankolkmeier