jankolkmeier/xbee-api #63: We would have a problem...

khinze, 2017-08-22:
I have two Xbee configured in Api mode 2. One connected rasberry pie configured as coordinator and one as the Router Api. 

The response I get on the Coordinator Serial monitor is as follows:
Receive Packet (API 1)

7E 00 C1 90 00 13 A2 00 40 E9 9C ED 22 C3 01 7B 22 47 54 59 50 22 3A 22 53 74 72 65 61 6D 22 2C 22 44 4F 22 3A 30 2E 30 2C 22 4C 54 4D 50 22 3A 30 2E 30 2C 22 45 43 22 3A 30 2E 30 2C 22 50 48 22 3A 30 2E 30 2C 22 53 54 45 50 5F 53 50 45 45 44 22 3A 31 30 2C 22 54 45 4D 50 31 22 3A 31 37 2E 33 35 2C 22 48 55 4D 49 44 31 22 3A 34 31 2E 37 33 2C 22 54 45 4D 50 32 22 3A 2D 31 30 30 30 2E 30 30 2C 22 48 55 4D 49 44 32 22 3A 2D 31 30 30 30 2E 30 30 2C 20 72 75 6E 74 69 6D 65 3A 33 31 31 30 34 31 2C 20 61 63 63 65 70 74 5F 63 6F 6D 6D 61 6E 64 73 3A 30 2C 20 22 47 45 4E 44 22 3A 30 7D 00 73

Start delimiter: 7E
Length: 00 C1 (193)
Frame type: 90 (Receive Packet)
64-bit source address: 00 13 A2 00 40 E9 9C ED
16-bit source address: 22 C3
Receive options: 01
RF data: 7B 22 47 54 59 50 22 3A 22 53 74 72 65 61 6D 22 2C 22 44 4F 22 3A 30 2E 30 2C 22 4C 54 4D 50 22 3A 30 2E 30 2C 22 45 43 22 3A 30 2E 30 2C 22 50 48 22 3A 30 2E 30 2C 22 53 54 45 50 5F 53 50 45 45 44 22 3A 31 30 2C 22 54 45 4D 50 31 22 3A 31 37 2E 33 35 2C 22 48 55 4D 49 44 31 22 3A 34 31 2E 37 33 2C 22 54 45 4D 50 32 22 3A 2D 31 30 30 30 2E 30 30 2C 22 48 55 4D 49 44 32 22 3A 2D 31 30 30 30 2E 30 30 2C 20 72 75 6E 74 69 6D 65 3A 33 31 31 30 34 31 2C 20 61 63 63 65 70 74 5F 63 6F 6D 6D 61 6E 64 73 3A 30 2C 20 22 47 45 4E 44 22 3A 30 7D 00
Checksum: 73

The response:
{"GTYP":"Stream","DO":0.0,"LTMP":0.0,"EC":0.0,"PH":0.0,"STEP_SPEED":10,"TEMP1":17.35,"HUMID1":41.73,"TEMP2":-1000.00,"HUMID2":-1000.00, runtime:311041, accept_commands:0, "GEND":0}

**Which mean the coordinator is receiving the packet correctly from the router connected to the ardiono**

**The error I get in the xbee api is as follows**
==> We would have a problem...
around line 170 in the xbee api. 

 if (S.buffer.length > S.offset) {
          S.buffer.writeUInt8(S.b, S.offset++);
        } else {
            console.log("We would have a problem...");
            S.waiting = true;
        }

**This is preventing me from being to able to read the package.**

How do I solve this problem when reading the data back in on my coordinator?

jankolkmeier, 2017-08-22:
Hey khinze.

Sorry for the cryptic logging message!
What's happening here is that the buffer allocated for reading messages is not big enough for the message you are sending.

In [lib/xbee-api.js#L39](https://github.com/jankolkmeier/xbee-api/blob/master/lib/xbee-api.js#L39), this buffer size is set. The default value is 128. If I recall correctly, this was chosen because it would be able to handle the maximum packet size of modules used at the time. However, doing a quick search on the internet, I find that there are other modules/firmwares that support more payload.

Can you change that number (i.e. to something like 512) to make sure this indeed solves your problem?

In the future, the buffer size should be configurable with an option!

samuelthf, 2017-10-10:
Hi jankolkmeier and khinze,
I had similar issues, and just like to point out that for it to work, I had to set value at [lib/xbee-api.js#L145](https://github.com/jankolkmeier/xbee-api/blob/master/lib/xbee-api.js#L145) to a bigger number too.
Thanks! Look forward to the future option.
- Samuel Tan

jankolkmeier, 2017-10-10:
Added the `parser_buffer_size` option and increased the default value so people won't run into this.

closed 2017-10-10 by jankolkmeier

maiorfi, 2017-10-17:
Hi. I think it should be edited line 146 of xbee-api.js, too, as follows:

```javascript
S.buffer = Buffer.alloc(this.options.parser_buffer_size);
```

jankolkmeier, 2017-10-17:
Thanks a lot, well spotted. While at it I also added a "builder_buffer_size" option, since if there are modules that can receive bigger payloads, they should also be able to send them.

khinze, 2017-10-23:
@jankolkmeier, Thanks. Much appreciated. I see you solved it in https://github.com/jankolkmeier/xbee-api/commit/eb99a6555ebe0583b64fc72d03e5805798921bd2.