jankolkmeier/xbee-api #65: xbee-stream module

exsilium, 2017-11-20:
Hi, noticed (and commented) on this commit: https://github.com/jankolkmeier/xbee-api/commit/0524b82141564b74ba5ed20abf34511dfa13b979

So where is xbee-stream being pulled in from? Where is it? It clearly blocks the [Travis tests](https://travis-ci.org/jankolkmeier/xbee-api/jobs/288980059) build. Why was this included in the release?

Cheers! 🍻

jankolkmeier, 2017-11-20:
I'm sorry, I was messing around with a fix for the new serial port version (didn't get there as easily as I thought) and accidentally left that line on the master branch. Thank you for reporting this!

closed 2017-11-20 by jankolkmeier

exsilium, 2017-11-20:
Thanks for the quick turnaround! I see 0.5.6 being published. Sweet! 🏆

commit 7d28039b (2017-11-20): fixes #65