jankolkmeier/xbee-api #69: Remote xbee sensor

MartinSokolov15, 2020-04-26:
Hi, can we connect remote xbee sensor (wall router), that is not connect to PC, to this xbee-api library and obtain data from the remote xbee sensor?

jankolkmeier, 2020-04-26:
This library's purpose is to create and parse 802.15.4/ZigBee/ZNet API frames. These frames might be read from/written to xbee devices connected to a PC through a USB/serial connection - but even the serial connection logic is not part of this module.

If you have problems with device compatibility, try the digi support forums: https://www.digi.com/support/forum/).

This library will also not simplify network configuration for you - i.e. if you can't make the devices talk to each other using the first party tools such as XCTU  (because of incompatibility or lack of understanding of the protocols), this library won't help you either. Again, if you need help with configuring your network, try the digi support forums.

Good luck!

closed 2020-04-26 by jankolkmeier

MartinSokolov15, 2020-04-26:
I don't have problems with devices compatibility, configuration or anything like that.
I just want to connect my remote sensors to node-red, in order to obtain data (temperature etc.) from them for my node-red app.

jankolkmeier, 2020-04-26:
Sorry if I sound snarky, but googling for "node-red xbee", gives you this: https://github.com/ramonmata/node-red-contrib-xbee - did try that?

Please make an effort on your own before asking for help. If you don't have an issue in particular, it is hard to help you. At least read the README's first, and provide all information in your original post right away.

MartinSokolov15, 2020-04-26:
Of course i try everything i found in node-red library, but every library example (like node-red-contrib-xbee) is looking for serial port, but the remote wireless sensors doesn't have one. So i can't connect them.

jankolkmeier, 2020-04-26:
You can't just connect to xbee devices without also connecting an xbee node to your PC.

If that's what you are doing, and your issue is that the xbee doesn't show up as a serial port, please ask on the digi forums for help.

As I wrote before, *and as stated in the README*, this library also needs to use a serial port to do the actual communication with xbee devices.

MartinSokolov15, 2020-04-27:
I wanted to connect remote xbee devices to the xbee node that is physically (serial port) connect to
PC or RPI, and obtain sensor data from the remote one (i know how to connect them with XCTU, but i want to use them with node-red, in order to create my app), but it seems like there is no such node-red library, correct?

The closest option is this one, but it's not exactly what I need.
https://flows.nodered.org/node/ncd-red-wireless

paolochiarlone, 2020-04-27:
As long as I know, officially there are only Arduino, C, C++, Java, Python libraries to talk to the XBee.
and yes you need a serial port (or SPI) locally to your PC

101100, 2020-04-27:
@MartinSokolov15 

If you have an xbee on your local machine and have set up the remote xbee with a node ID (or have its serial number which doubles as its address) then you should be able to use the remote AT command to work with the remote xbee, including reading from sensors or setting up polling sensor packets from the module.

I have made a library that connects to a local xbee called [xbee-rx](https://github.com/101100/xbee-rx) based on `xbee-api`.  I have an example script to send a remote AT command [here](https://github.com/101100/xbee-rx/blob/master/examples/simple-remote.js) that I have used myself to send a command to an xbee elsewhere in my house from an xbee connected to my computer.  Another script I was using to get regular temperature sensor readings remotely is [here](https://github.com/101100/xbee-rx/blob/master/examples/temperature.js).