jankolkmeier/xbee-api #70: UnhandledPromiseRejectionWarning: Error

Rezcue, 2021-03-27:
Hi
My system is running and sometimes i get this error. I cant find the reason for this.
I had the same problem some days ago. I thought i solved it but i guess i was wrong.

(node:21369) UnhandledPromiseRejectionWarning: Error: Checksum Mismatch {"buffer":{"type":"Buffer","data":[126,0,15,144,0,19,162,0,65,195,131,30,160,156,1,49,51,53,127,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]},"offset":19,"length":15,"total":1216,"checksum":127,"b":127,"escape_next":false,"waiting":false}
    at XBeeAPI.parseRaw (/home/pi/.node-red/node_modules/xbee-api/lib/xbee-api.js:220:19)
    at Transform.XBeeAPI.parser._transform (/home/pi/.node-red/node_modules/xbee-api/lib/xbee-api.js:48:19)
    at Transform._read (_stream_transform.js:191:10)
    at Transform._write (_stream_transform.js:179:12)
    at doWrite (_stream_writable.js:403:12)
    at writeOrBuffer (_stream_writable.js:387:5)
    at Transform.Writable.write (_stream_writable.js:318:11)
    at SerialPort.ondata (_stream_readable.js:718:22)
    at SerialPort.emit (events.js:314:20)
    at addChunk (_stream_readable.js:297:12)
(node:21369) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:21369) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

**Im using a Ultrasonic Distancesensor HCSR04.**

`#include <SoftwareSerial.h>
#include <HCSR04.h>
#include <XBee.h>

XBee xbee = XBee();
SoftwareSerial my(4,2); // TX, RX
int triggerPin = 13;
int echoPin = 12;
UltraSonicDistanceSensor distanceSensor(triggerPin, echoPin);

char Buffer[255];
// SH + SL Address of receiving XBee
XBeeAddress64 addr64 = XBeeAddress64(0x0013A200, 0x41C38580);

void setup () {
     Serial.begin(115200);
     my.begin(115200);
     xbee.setSerial(my);
     Serial.println("System Up!");
}

void loop () {
  
    
    int distance = distanceSensor.measureDistanceCm();
    String d = String(distance);
    d.toCharArray(Buffer, 255);
    ZBTxRequest zbTx = ZBTxRequest(addr64, (uint8_t *)Buffer, strlen(Buffer));
    Serial.println(d);
    xbee.send(zbTx);
    delay(5000);
       
}`

And my Settings for my Xbee are:

https://gyazo.com/b0d84a56666f9f477285149939ec4fef

Can someone hepl me out here?

Rezcue, 2021-03-27:
I did a work around to this. -->
https://discourse.nodered.org/t/node-red-node-modules-xbee-api-lib-xbee-api-error/43039

jankolkmeier, 2021-03-27:
I don't have time to exactly examine what's wrong with the frame you've posted further up, but here are some thoughts for now:

- I don't know where / how the node red module is initializing the xbee api object, but make sure if it is using API mode 2  instead of 1 (see the api_mode property in the options). This needs to match your xbee modules API mode (which uses AP=2 based on the screenshot).
- I'm not familiar with ZBTxRequest/the arduino code here, but I wouldn't use ```Serial.println()``` to write API frames - unless I'm missing something, this will cause an unnessecary newline character to be written after the frame. Just use ```Serial.print()``` or ```Serial.write()```
- I'm hoping ZBTxRequest doesn't expect you to escape the data if operating in AP=2 mode (I really doubt it, but it might be something to look into, just to be sure)
- Probably not the issue here, but as general advice, I wouldn't recommend converting the int object to string. You should be able to just ```ZBTxRequest(addr64, (uint8_t *) distance, sizeof(distance))```, which will write the 4 bytes representing the int to the body. You can then convert the four bytes to an int object on the other end (i.e. for javascript, use the ```buffer-reader``` module to wrap around the binary data you receive and run ```nextInt32LE()``` to read an arduino integer).

Rezcue, 2021-03-27:
Hey

1. Xbees are on API 2
2. Serial.println() should only print it in the serial monitor (arudino) only xbee.send(zbTx); sends it over XBee
3. No idea what that means im new to this 
4. I had it first on Int but i got that error pretty much every time i got a value over 250cm. Some dude recommened to convert the Int in String. And it actually helpen a lot. I think int 250 is to much on bytes for the xbee to handle. The xbee can handle max 255 bytes. That error did not appear that often anymore but some times it still does. 

https://www.digi.com/support/knowledge-base/maximum-payload-size-for-digi-rf-products

I can recreate that error fast when i block one of the ultrasonicoutputs of the senor. So that the sensor gives out a -1 (for error i think). After some loops that error pops up.

Sry if my english sucks :D im trying my best

jankolkmeier, 2021-03-28:
1. I know your xbees are on API 2, I saw in the screenshot. but is the node-red module initializing xbee-api with api mode 2 as well?
2. Oops sorry, I overlooked that!
3. Goes back to the issue with API mode mismatch. Maybe [start here](https://www.digi.com/resources/documentation/Digidocs/90001942-13/concepts) with learning about it.  /c_api_escaped_operating_mode.htm?TocPath=XBee%20API%20mode%7COperating%20mode%20configuration%7C_____1)
4. If you're referring to [this](https://discourse.nodered.org/t/node-red-node-modules-xbee-api-lib-xbee-api-error/43039/9), that's not at all how you would send an int. You're trying to put an int (which is 4 bytes) into your "payload" object of 1 bytes. I.e. you're not sending 3/4 of the bytes that make up that int.

If you're forming the packets properly on the arduino end, you're not going reaching the payload limit when sending it as a string (unless your string is 94+ characters long, which is a very big number, or when talking fractionals, a number with a lot of positions after the dot). And you're certainly not going to hit the payload limit if you're sending the number in binary properly (you should be able to fit the number at least 23 times in a single packet).

Rezcue, 2021-03-28:
1. "node-red module initializing xbee-api with api mode 2" What is that module and how can i see the configs of it? 
4. Unfortunately the skills i have in programming are those of a newbie. I have no idea how to do it right with Ints. 

Im quite happy now that my system is running. It was really annoying that it did not restart after the error but as long as it does restart im fine with that. Its now running 2 days without the error.

Thank u for your time and help on this matter!

tatujan, 2021-05-04:
I have the same error, but I am not using node-red. I am simply trying to parse API 2 (0x13 escaped) frames coming from my PI's serial port. Even though I pass correctly escaped byte data from my serial port to form xbee frames, I get this checksum mismatch error. I've converted the error log's decimal array to hex, and it looks like xbee-api library converts it to a non-escaped frame, and everything crashes.

here are my logs: 

`(node:3880) UnhandledPromiseRejectionWarning: Error: Checksum Mismatch {"buffer":{"type":"Buffer","data":[126,0,104,144,0,19,162,0,65,209,22,240,255,254,194,10,3,71,65,77,18,32,57,54,50,51,99,50,57,56,57,50,51,101,52,102,56,52,97,49,56,51,55,49,51,53,97,98,53,101,57,57,100,98,26,24,65,78,68,82,79,73,68,45,57,53,52,55,48,49,55,51,102,49,55,54,99,56,97,102,34,23,65,78,68,82,79,73,68,45,51,53,55,208,229,186,130,194,162,186,194,170,194,154,194,66,2,24,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]},"offset":108,"length":104,"total":8642,"checksum":24,"b":24,"escape_next":false,"waiting":false}
    at XBeeAPI.parseRaw (/home/pi/dev/GatewayNodeJS-0.1/node_modules/xbee-api/lib/xbee-api.js:220:19)
    at Transform.XBeeAPI.parser._transform (/home/pi/dev/GatewayNodeJS-0.1/node_modules/xbee-api/lib/xbee-api.js:48:19)
    at Transform._read (_stream_transform.js:190:10)
    at Transform._write (_stream_transform.js:178:12)
    at doWrite (_stream_writable.js:415:12)
    at writeOrBuffer (_stream_writable.js:399:5)
    at Transform.Writable.write (_stream_writable.js:299:11)
    at SerialPort.ondata (_stream_readable.js:710:20)
    at SerialPort.emit (events.js:198:13)
    at addChunk (_stream_readable.js:288:12)
(node:3880) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:3880) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
`

Here is how that decimal data looks like in hex. Attention to '13' as it should be '7D 33':

`7E 00 68 90 00 13 A2 00 41 D1 16 F0 FF FE C2 0A 03 47 41 4D 12 20 39 36 32 33 63 32 39 38 39 32 33 65 34 66 38 34 61 31 38 33 37 31 33 35 61 62 35 65 39 39 64 62 1A 18 41 4E 44 52 4F 49 44 2D 39 35 34 37 30 31 37 33 66 31 37 36 63 38 61 66 22 17 41 4E 44 52 4F 49 44 2D 33 35 37 D0 E5 BA 82 C2 A2 BA C2 AA C2 9A C2 42 02 18`

any idea?

jankolkmeier, 2021-05-04:
Hi tatujan, thanks for bringing this up. Would it be possible for you to set up a minimal working (or rather failing) example? Just to confirm that you're configuring xbee-api correctly? Alternatively/in addition, could you maybe set up a test case with the problematic frame (see [xbee-api_test.js#L549](https://github.com/jankolkmeier/xbee-api/blob/master/test/xbee-api_test.js#L549) for some existing test cases with an api_mode: 2 frame)?

tatujan, 2021-05-04:
Hi jankolkmeier, I have implemented a simple serial port listener where XBee passes the incoming packages and xbee-api parses the buffer. I have two different setups and it shows up in only one. I have used different xbees and computers (Pi), this issue pops up every now and then. I am logging incoming bytes and sometimes it does not match what has been received on the XBee, xctu even cannot form a frame out of it sometimes. So this could be a hardware issue as well. I am still investigating and have not found a definitive issue yet. I'll keep you posted.

jankolkmeier, 2021-05-05:
Sure, keep me posted.

Going back to your previous message: what did you mean by ```Attention to '13' as it should be '7D 33'```?
Just to be clear, if it was '7D 33' in the raw data read from the serialport, then it is correct that it should come out as 13 (this is the escaping in action).  

And in case it also wasn't clear where this error is stemming from in general: The exception is caused when there is a checksum mismatch in the package. The checksum in your sample package is 0x18 = 24 (dec), however, the checksum of the frame is 8642 % 256 = 194 (dec). Since these are not equal, there is a real possibility that this is an actual incorrectly transmitted package - i.e. there are bytes that got corrupted in transmission, and thanks to the checksum we were able to catch this.

To debug this further, you could parse the data from the xbee regardless, ignorning the incorrect checksum. Then you check if it is really what you've passed to the sending xbee on the other end before.

 I've done this for your sample package real quick:
```♥GAM↕ 9623c298923e4f84a1837135ab5e99db→↑ANDROID-95470173f176c8af"↨ANDROID-357ÐåºÂ¢ºÂªÂÂB☻```

There are some funny characters in there. I don't know what your application layer protocol looks lik -, the mix of binary and ascii data might be intentional - but it could also be genuinely corrupt data! In that case you should be happy to know that xbee-api has caught this for you ;)

tatujan, 2021-05-06:
Hi jankolkmeier,

Yes, I am knowledgeable in XBee frames, checksums, and escape characters. I guess I didn't know the library was taking the escaped frame and forming a non-escaped one for parsing. 

The data in question is partially intact and partially corrupted. I have investigated the issue immensely over the last 2-3 days. I was suspecting the following components:

- SerialPort is not passing correct data
- XBee API has a bug
- XBee radio module itself is acting up.

I have discarded the XBee radio because I've tried with 3 different XBee radios and development boards at the time. After lots of tests, I also discarded the XBee API bug, because the incoming data was already altered and was not valid. I had the wrong frame length bytes and overall corrupted data. At this point, I was sure something was up with the serial port. I have seen that on Raspberry Pi 3s and 3Bs, having the Bluetooth module activated may result in some serial port byte corruption or not receiving any data altogether. So I gave that a try. However, disabling Bluetooth did not fix the issue. So I was back to square one. In the meantime, I got a new radio and development board just to make sure it is not hardware and voila. With the new board and radio, everything worked perfectly. I have checked my 3 radio modules with the new board I got. They also worked fine. So I came to the conclusion that the 3 boards I have, gone bad simultaneously made me think suspect other components. Overall after a very painful debugging, the issue was with the development boards. 

Thank you for your swift responses!

jankolkmeier, 2021-05-07:
Thanks for the explanation, yes - sounds like a challenging issue to track down. Happy to hear it works for you now, and good luck with your project going forward :)

closed 2021-05-07 by jankolkmeier